ataradov/usb-sniffer #1: Request - JLPCB/LCSC BOM & CPL

hanzov69, 2023-06-14:
For the less experienced folks (like myself) out there, would it be possible to provide a BOM & CPL suitable for use with a service like JLCPCB (which is integrated with LCSC and EasyEDA)?

ataradov, 2023-06-14:
Creating BOM should not be that hard, but none of the components have pick points defined. I think most of them are just in the geometric center of the part. LCSC also seems to have most of the major parts. They don't have comparators (well, they have 2 in stock).

I have never ordered board assembly so, I don't know what the requirements are. And I would not be comfortable publishing something I did not test myself. If someone creates the necessary files and tests the whole process with JlcPcb, I can add them to the repo for others to use.

ataradov, 2023-06-14:
But also, if they only assemble boards from EasyEDA and not just random boards, then I don't think there is much we can do. I'm not going to redo the design in EasyEDA for sure.

hanzov69, 2023-06-15:
> I would not be comfortable publishing something I did not test myself
Totally respect that!

> I'm not going to redo the design in EasyEDA for sure.
Naturally. They support pulling in gerber files, so that's not a requirement at all. 

EasyEDA/JLCPCB/LCSC are all the same company, so they have an "integrated vertical" type pipeline, but you're not limited to doing it that way. JLCPCB will spin the board just fine with what you've provided, it's only if you want to do assembly with them that you'll need a bom/cpl.

I spent some time after opening this issue, using this plugin ( https://github.com/Bouni/kicad-jlcpcb-tools ) with KiCad I was able to _mostly_ generate a BOM/Placement file for JLCPCB. Biggest thing I ran in to is my own limited knowledge- Looks like LCSC has zero availability on some components and I'm not well enough versed to know what component alternatives (if any) I can use. 

I may update the footprints with the exact LCSC part #'s to help along others going down this rabbit hole. If I get something suitable for doing assembly with JLCPCB (and/or EasyEDA formatted), I'll fork the repo and ping (since I respect not wanting to include things you haven't tested).
Thanks for the response!

closed 2023-06-15 by hanzov69

ataradov, 2023-06-15:
What components are not available? I can help with equivalents. I checked all the major ICs and they were available (although FPGA is much more expensive than on mouser). The only non-generic thing that is not available are comparators. They only had two. 

For the clock generators, anything of the specified frequency would work, they don't have to be Microchip parts.

The rest are generic components.

If someone wants to go though the JlcPcb ordering process and test the result, I'm available for help. I'd like people to be able to get the hardware, I just don't have the bandwidth  to do that myself.

sugoku, 2023-06-16:
Hello, I just went through the JLCPCB process myself after finding this and the only things I wouldn't be able to get are the 512 ohm resistor (unless it's supposed to be 5.1K) and the USB3343 (which is out of stock, currently). The USB3343 can be sourced from LCSC but I would have to solder it myself. Also, is there a required size for the EEPROM? I chose a 64Kbit chip.

ataradov, 2023-06-16:
512 Ohm is supposed to be 510 - a very common value. The value of this resistor does not matter, it just provides the current limiting for the trigger input. Anything in the range 0-5 kOhm would work.

64 kBit would work, but I updated the schematic this morning to show 24LC128 - the maximum that can be supported.

Clockwork-Muse, 2024-03-06:
@hanzov69 - Hey, do you have a sample/starting file you can share?  I'm looking at maybe ordering some as well.