ataradov/usb-sniffer #5: [BUG]usb-sniffer software cannot write firmware unless sudo in linux

huajijam, 2023-06-22:
In windows10(22H2 19045.3086),after driver loaded, 
`./usb_sniffer --mcu-sram usb_sniffer.bin` cannot write firmware

got: **Error: libusb_open(): LIBUSB_ERROR_NOT_SUPPORTED**

with elevate(run as admin), still got same error
[image]

in linux,got: **Error: libusb_open(): LIBUSB_ERROR_NOT_SUPPORTED**
but sudo will work
[image]

ataradov, 2023-06-22:
For Windows - can you show how it appears in the Device Manager? Did you actually install the INF file?
Does it work after you program the EEPROM fro Linux?

For Linux, this is also expected. For it to work under normal user you need to add the file /etc/udev/rules.d/90-fx2.rules  with the following contents:

ATTRS{idVendor}=="04b4", ATTRS{idProduct}=="8613", MODE="0666"

Re-plug the device and it should get proper permissions. You may need to add one more line for 6666:6620 with the same mode for normal operation.

I'll document this a bit later and provide full files.

ataradov, 2023-06-23:
I've added udev rules file https://github.com/ataradov/usb-sniffer/blob/main/bin/90-usb-sniffer.rules . Copy that into /etc/udev/rules.d/ and Linux permissions would be sorted out.

For the Windows side - this behavior means that INF file was not installed. It is necessary for the blank device only. If you program EERPOM first time under Linux, device should be  recognized by Windows automatically.

huajijam, 2023-06-23:
> I've added udev rules file https://github.com/ataradov/usb-sniffer/blob/main/bin/90-usb-sniffer.rules . Copy that into /etc/udev/rules.d/ and Linux permissions would be sorted out.
>
> For the Windows [...]

I found out that during the driver installation,there is a error

[image]

is your driver not signed?

ataradov, 2023-06-23:
This is not a "driver" really, it is a dummy INF file that installs a standard generic WinUSB driver from Windows. This INF file was created automatically by Zadig tool. I have no clue what is necessary for Windows to consider it "signed". I've included a catalog file (blank_fx2lp.cat), which  I think related to that. 

This INF file installed fine on my test Windows 10 PC. No idea what is different, windows just sucks. 

You can download Zadig tool (https://zadig.akeo.ie/) and make your own "drivers". They may work better. Select your device from the list and select a WinUSB driver.

huajijam, 2023-06-23:
> This is not a "driver" really, it is a dummy INF file that installs a standard generic WinUSB driver from Windows. This INF file was created automatically by Zadig tool. I have no clue what is [...]

if the driver is from zadig,then let user use zadig to install,it will work

huajijam, 2023-06-23:
> This is not a "driver" really, it is a dummy INF file that installs a standard generic WinUSB driver from Windows. This INF file was created automatically by Zadig tool. I have no clue what is [...]

and yes,problem because driver is not signed

[image]

ataradov, 2023-06-23:
I'm not sure what the difference is, I tested that on my Win10 PC and it worked. I'm not sure is there is Pro vs Home or something like that. I don't use Windows.

I though catalog file contained the signature (as confirmed by this https://learn.microsoft.com/en-us/windows-hardware/drivers/install/catalog-files). Do you have it present in the same directory as INF file?

huajijam, 2023-06-23:
> I'm not sure what the difference is, I tested that on my Win10 PC and it worked. I'm not sure is there is Pro vs Home or something like that. I don't use Windows.
>
> I though catalog file contained [...]

I just using usb-sniffer\bin folder as the driver folder

this is the fix and you might do it before,and mine might just been disable by windows again
https://thegeekpage.com/windows-encountered-a-problem-installing-the-driver-software/

ataradov, 2023-06-23:
It might have been enabled by default. I don't remember doing anything like this before. The Windows PC I use for testing was purchased just for testing of stuff like this because I did not have any Windows PCs. I have not used it for 5+ years and updated it to the latest version just to test this driver. I also used full bin directory and it just silently worked.

If Zadig does not supply sufficiently signed drivers, there is not much I can do.

This step is needed just for the initial programming. Once the real firmware runs, it tells the OS to install the same WinUSB driver though the MSOSv2 descriptor requested over USB on the first connection.

This also highlights the issue with this signing thing - any USB device that you plug into the PC can install that driver automatically without user involvement at all.

closed 2023-06-23 by huajijam

Notedop, 2024-09-10:
I just had the same issue on Windows 11, I could not install the initial driver with the supplied INF file. When using Zadig directly it works, it might be worth making this note in the instructions as I had to search the tickets to get to this solution.

stappersg, 2024-09-11:
On Tue, Sep 10, 2024 at 01:12:50PM -0700, Notedop wrote:
> it might be worth making this note in the instructions
> as I had to search the tickets to get to this solution.

"Patches welcome" and do know that a poor merge request
is much better as no merge request