ataradov/usb-sniffer #13: I used 4TG100C, Transfer rate:  0.00 MB/s

vgegok, 2023-07-20:
I used diamonds to regenerate the jet file of 4TG100C, and initially everything was normal. However, when I tried to measure the speed, I got a result of 0 on Windows, and Linux would directly report an error.

# Windows:

```
PS E:\USB\usb-sniffer-main\usb-sniffer-main\bin> .\usb_sniffer_win.exe --mcu-sram .\usb_sniffer.bin
Uploading 3187 bytes into the FX2LP SRAM
...done

PS E:\USB\usb-sniffer-main\usb-sniffer-main\bin> .\usb_sniffer_win.exe --mcu-eeprom .\usb_sniffer.bin
Programming 3187 bytes into the FX2LP EEPROM (SN: 443602XXXXXXXX)
...done

PS E:\USB\usb-sniffer-main\usb-sniffer-main\bin> .\usb_sniffer_win.exe --fpga-flash ..\fpga\impl\usb_sniffer_impl.jed
Programming FPGA flash
Erasing flash
Programming configuration data .....
Verifying configuration data
Programming and verifying Feature Row
Programming and verifying FEABITS
Exiting programming mode
...done

PS E:\USB\usb-sniffer-main\usb-sniffer-main\bin> .\usb_sniffer_win.exe --test
Starting speed test
Transfer rate:  0.00 MB/s
Transfer rate:  0.00 MB/s
Transfer rate:  0.00 MB/s
Transfer rate:  0.00 MB/s
Transfer rate:  0.00 MB/s
Transfer rate:  0.00 MB/s
Transfer rate:  0.00 MB/s
Transfer rate:  0.00 MB/s
Transfer rate:  0.00 MB/s
Transfer rate:  0.00 MB/s
```

# Ubuntu 20.04:

```
$ ./usb_sniffer --test
Starting speed test
Error: usb_capture_callback(): 4
```

ataradov, 2023-07-20:
Callback status 4 is LIBUSB_TRANSFER_STALL. This is strange,because FX2LP should not stall on hardware endpoints. What was the source for the FX2LP device?

From the FPGA side, I would verify that PHY oscillator is running at 26 MHz and that there is 60 MHz clock on the output from the PHY.

Speed grade 4 device does not meet timing requirements, so you should see warnings about that in the log. It might work at room temperature, I guess.

vgegok, 2023-07-20:
Your answer gave me a hint, I know why. Since the 8.06K resistor has not yet arrived, I did not weld it! 

I haven't read your code carefully yet. 
At first, I didn't know which clock source the FPGA uses. I thought that the USB3343 wouldn't affect speed testing if it didn't work. 
Now I know that the FPGA clock is provided by USB3343 (60MHz), so when conducting speed testing, the 8.06K resistor must be soldered. 
Looking forward to receiving it! 

Thank you again for your answer!

---- Replied Message ----
| From | Alex ***@***.***> |
| Date | 07/20/2023 23:12 |
| To | ataradov/usb-sniffer ***@***.***> |
| Cc | vgegok ***@***.***>,
Author ***@***.***> |
| Subject | Re: [ataradov/usb-sniffer] I used 4TG100C, Transfer rate: 0.00 MB/s (Issue #13) |

Callback status 4 is LIBUSB_TRANSFER_STALL. This is strange,because FX2LP should not stall on hardware endpoints. What was the source for the FX2LP device?

From the FPGA side, I would verify that PHY oscillator is running at 26 MHz and that there is 60 MHz clock on the output from the PHY.

Speed grade 4 device does not meet timing requirements, so you should see warnings about that in the log. It might work at room temperature, I guess.

vgegok, 2023-07-21:
I soldered an 8.2K resistor at position R11, and now the test is working properly!

At first, I tested the clock signals of 26MHz (PIN21) and 60MHz (PIN2) with an oscilloscope and found that 26MHz was normal, but there was no 60MHz clock. After I welded the resistor, everything was normal.

As I said, this bias resistor will affect the operation of USB3343, resulting in the inability to output a 60MHz clock signal.

```
 .\usb_sniffer_win.exe --test
Starting speed test
Transfer rate: 48.57 MB/s
Transfer rate: 48.96 MB/s
Transfer rate: 48.71 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.71 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.96 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.67 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.96 MB/s
Transfer rate: 48.71 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.67 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.96 MB/s
Transfer rate: 48.71 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.91 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.86 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.67 MB/s
Transfer rate: 48.76 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.62 MB/s
Transfer rate: 48.71 MB/s
Transfer rate: 48.81 MB/s
Transfer rate: 48.62 MB/s
Transfer rate: 48.71 MB/s
```

vgegok, 2023-07-21:
@ataradov, Thank you.
I will close this issues.

closed 2023-07-21 by vgegok

ataradov, 2023-07-21:
Good to know that it actually does not even start without that resistor.