ataradov/usb-sniffer #14: Error: libusb_bulk_transfer(): LIBUSB_ERROR_OVERFLOW, usb_capture_callback(): 6

croofec, 2023-07-21:
I can't run a speed test after successfully installing. The connection between the USB and the computer works fine, I can't sniff
Current usblib version: libusb-1.0.26, all parts from mouser.

[image]

[image]

Hardware or software issue?

ataradov, 2023-07-21:
This may be related to a recent change. 

Build the binary from the source and change TRANSFER_COUNT in usb.c from 16 to 8 or even 4.

If this helps, please check what is the maximum value that causes the issue.

I'm not sure what defines how many transfers libusb would accept at the time. It appears to be system dependent.

croofec, 2023-07-24:
Changing the parameter had no effect even when I set it to 1.  I started debugging the software and always the bulk transfer buffer returned the size of 64 which is smaller than the size of 512. This could not cause overflow.  But in the dmesg:

[image]

I changed the slot from USB2.0 to USB3.0 and it works :D

closed 2023-07-24 by croofec

ataradov, 2023-07-24:
Ok, yes, I don't do anything to accommodate the case when it enumerates as a Full-Speed device.

HS mode is USB 2.0, so there must be something wrong with that specific port or your cables. It looks like it tried to enumerate as HS device, but could not.