ataradov/usb-sniffer #18: Only syslog entries

brunopp, 2023-08-16:
I just assembled the board and followed the instructions to flash the device and get the desired speed of 40-50 MB/s - all good.

When connecting a usb device (mouse -> full speed) to the port I get data in wireshark, but no usb data, I only get "Syslog" entries. I have also tried different usb drives at high speed. Also the log stops after about about 1000 entries (only for high speed). I have selected the correct Capture Speed in wireshark.

All the devices I have tried works when connected through the usb-sniffer board.

I have also noticed I get VBUS OFF in one of the first entries as opposed to VBUS ON the examples under /doc.

Logs: [wireshark.zip](https://github.com/ataradov/usb-sniffer/files/12356815/wireshark.zip)

Also only two LED's light up (LED 1 and LED 7), are the other LED supposed to light up?

As for the BOM I could not find the exact oscillators, but found others within 50 ppm:
[DSC1001CI2-024.0000](https://www.mouser.dk/ProductDetail/998-1001CI2-024.0000)
[DSC1001CI5-026.0000](https://www.mouser.dk/ProductDetail/998-1001CI5-026.0000)

ataradov, 2023-08-16:
This looks like there is some issue with the USB PHY. 

First of all, check that the soldering looks ok. Attach magnified pictures, if you can.

Then check that you have 26 MHz on the output of the oscillator and 60 MHz on the output of the PHY (CLKOUT pin).

If you look at the HS log, you will see a lot of JK toggles, which start with "[singel]" and continue with "[both]". This is a normal part of the HS speed negotiation. What is wrong here is that they are supposed to be ~50 us in duration and in your case they are 20 us. This looks like incorrect clock is being used.

ataradov, 2023-08-16:
And LEDs are fine, the 4 other LEDs are there for debugging and are not used in the release version.

brunopp, 2023-08-16:
The oscillator (IC6) is outputting 26 MHz. But I'm not getting 60 MHz on the USB (IC4) on pin 2, but 26 MHz as well as shown below:

[Circuit](https://github.com/ataradov/usb-sniffer/assets/6736551/7c2ae64c-b328-4abf-bd88-24536002b0bd)
[Oscilloscope (IC4, pin 2)](https://github.com/ataradov/usb-sniffer/assets/6736551/7dd35c75-c68a-4e8f-8072-c623b71c555f)

Which is odd? The according to the [datasheet](https://eu.mouser.com/ProductDetail/Microchip-Technology/USB3343-CP-TR?qs=pA5MXup5wxEezmLOdrNXbw%3D%3D&countryCode=DK&currencyCode=DKK) it should output 60 MHz.

ataradov, 2023-08-16:
What is the value of the resistor on the pin 19 (RBIAS). It is critical that it is 8.06 kOhm +/-5%.

Neither B22 or 822 seem to match that value, but SMD codes are all over the place. So check the value with a multimeter.

brunopp, 2023-08-16:
It isn't clear how important exact values are from the BOM, so I just used the closest I had 8.2kOhm +-1%. I tried to solder 470k in parallel with 8.02k getting 8.06k measured (isolated from the circuit), but no luck.

Are there other critical resistor values in the circuit?

ataradov, 2023-08-16:
If the value is specified with that accuracy, it must be important. If 8.2 kOhm worked, it would have been on the schematic, it is not like I'm adding random values for fun.

No other values are critical as far as I know, so I would look for possible shorts.

Or may be reflow or change the PHY and see if that helps. I have never seen it output the input clock directly. Usually it is either nothing if the resistor is missing entirely, or 60 MHz.

ataradov, 2023-08-16:
And check for other voltages too, like VDD18 should be at 1.8 as supplied by the internal voltage regulator. If it is not at 1.8, check that C15 is placed and has correct value.

closed 2023-08-16 by brunopp

ataradov, 2023-08-16:
The JK chirp at the beginning still looks messy. It was way better in your original capture. It should be clean J/K toggles at ~50 us internal.

Double check your resistor divider ladder for the compactors.

brunopp, 2023-08-16:
Strange, I didn't reflow the backside where the resistor dividers are placed nor the comparators on the top side.

ataradov, 2023-08-16:
It feels like something is not soldered fully. There should be no random line state changes in the middle of the data.

I would carefully check all the comparators and related tracks pins. May be measure reference voltage levels on the resistive ladder and make sure that they are correct.

brunopp, 2023-08-16:
You are talking about the high speed log? Specifically: Syslog Line state: undefined? For example line no. 7715 and 8893.

I'll try some additional reflow.

ataradov, 2023-08-16:
Yes. Look at your original HS log. At the beginning you can see that there are K/J state pairs, first [single] and then [both]. Their timing is not correct because it was running at 26 MHz, but the order is fine. 

The time in the original log is 21 us, but the actual real time is (60/26) * 21 us = 48.5 us. This is the time you should expect.

You should see the same in the new log. There should not be undefined states. It is fine if there are some due to transient, but they should not be long.

brunopp, 2023-08-16:
Reflowing the comparators and divider resistors seemed to have worked, I find no undefined states (except a little bit at the beginning, which should be due to transient as you mentioned).

Logs: [wireshark - after reflow.zip](https://github.com/ataradov/usb-sniffer/files/12364141/wireshark.-.after.reflow.zip)

ataradov, 2023-08-16:
Yep, this is the correct result.