ataradov/usb-sniffer #22: Are there any barriers to a MacOS version?

robertrau, 2023-09-17:
First, thank you for this project, this required a lot of research and work.
I see Infineon supports MacOS with the CY7C68013A MCU. You have done the heavy lifting on the FPGA, so it is a bitstream load and that seems to be through your firmware.
Are there USB requirements that are unique to MacOS?
I would just like to use my Mac with I am sniffing a connection to my Windows machine.

Thanks again for all the hard work.

Bob

ataradov, 2023-09-17:
No, there are not. I don't have any Apple hardware, so I have no way to make a port. Assuming libusb works there, the port should not be too hard, but I'm not going to do it and maintain it. You can try to build it from the source and see what happens. The Linux version would be the closest, but I expect there to be some minor OS-specific issues.

I don't use anything from the Infineon, so whatever they have is irrelevant here.

robertrau, 2023-09-17:
Thank you for the quick answer. For now I'll be happy with the Windows version. If I get a MacOS version running I will upload it.

fizzyade, 2023-11-21:
it builds and works fine under macOS, I simply "hacked" the Makefile to work on my system rather than do it properly since paths need to be added to find the package configs for libraries provided via brew, since I used libusb it was already present in my system via brew, so I just linked directly to it and added the header path to the includes,

`
CFLAGS += -W -Wall --std=gnu11 -O3
CFLAGS += -D_GNU_SOURCE
CFLAGS +=  -I/usr/local/Cellar/libusb/1.0.26/include/libusb-1.0

SRCS = \
  usb.c \
  fpga.c \
  fx2lp.c \
  capture.c \
  os_common.c \
  usb_sniffer.c \

HDRS = \
  usb.h \
  fpga.h \
  fx2lp.h \
  capture.h \
  os_common.h \
  usb_sniffer.h \

UNAME := $(shell uname)

ifeq ($(UNAME), Darwin)
  BIN = usb_sniffer
  CFLAGS += -DOS_LINUX
  LDFLAGS += -L/usr/local/Cellar/libusb/1.0.26/lib
  LDFLAGS += -lusb-1.0
  EXTCAP_PATH = ~/.config/wireshark/extcap
else
  BIN = usb_sniffer.exe
  CFLAGS += -DOS_WINDOWS
  LDFLAGS += -Wl,--subsystem,console
  LIB_PATH = `pkg-config --variable=libdir libusb-1.0`
  LDFLAGS += $(LIB_PATH)/libusb-1.0.a
  LDFLAGS += $(LIB_PATH)/libwinpthread.a
  EXTCAP_PATH = $(APPDATA)/Wireshark/extcap/
endif

all: $(BIN)

$(BIN): $(SRCS) $(HDRS)
	gcc $(CFLAGS) $(SRCS) -o $(BIN) $(LDFLAGS)

clean:
	rm -fr $(BIN)

install: $(BIN)
	cp $(BIN) $(EXTCAP_PATH)

prog_eeprom:
	./usb_sniffer --mcu-eeprom ../firmware/usb_sniffer.bin

prog_sram:
	./usb_sniffer --fpga-sram ../fpga/impl/usb_sniffer_impl.bit

prog_flash:
	./usb_sniffer --fpga-flash ../fpga/impl/usb_sniffer_impl.jed
`

fizzyade, 2023-11-22:
I removed the hardcoded paths for macOS in my hack, this now uses brew to discover the location of it's pkg-config and libusb, so....

you need to have home-brew installed, and then run these commands before building the firmware:

```bash
brew install pkg-config
brew install libusb
```

then you can do 'make' and providing the required brew libraries have been installed, the code will compile and a resulting usb_sniffer binary will be built.

```Makefile
CFLAGS += -W -Wall --std=gnu11 -O3
CFLAGS += -D_GNU_SOURCE

UNAME := $(shell uname)

ifneq ($(UNAME), Darwin)
  CFLAGS += -fno-diagnostics-show-caret
  CFLAGS += `pkg-config --cflags libusb-1.0`
endif

SRCS = \
  usb.c \
  fpga.c \
  fx2lp.c \
  capture.c \
  os_common.c \
  usb_sniffer.c \

HDRS = \
  usb.h \
  fpga.h \
  fx2lp.h \
  capture.h \
  os_common.h \
  usb_sniffer.h \

ifeq ($(UNAME), Linux)
  BIN = usb_sniffer
  CFLAGS += -DOS_LINUX
  LDFLAGS += -lm -lrt
  LDFLAGS += `pkg-config --libs libusb-1.0`
  EXTCAP_PATH = ~/.config/wireshark/extcap
else ifeq ($(UNAME), Darwin)
  BREW_PKGCONFIG := $(shell brew --cellar pkg-config)/$(shell brew list --versions pkg-config | tr ' ' '\n' | tail -1)/bin/pkg-config
  BIN = usb_sniffer
  CFLAGS += -DOS_LINUX
  CFLAGS += `$(BREW_PKGCONFIG) --cflags libusb-1.0`
  LDFLAGS += `$(BREW_PKGCONFIG) --libs libusb-1.0`
  EXTCAP_PATH = ~/.config/wireshark/extcap
else
  BIN = usb_sniffer.exe
  CFLAGS += -DOS_WINDOWS
  LDFLAGS += -Wl,--subsystem,console
  LIB_PATH = `pkg-config --variable=libdir libusb-1.0`
  LDFLAGS += $(LIB_PATH)/libusb-1.0.a
  LDFLAGS += $(LIB_PATH)/libwinpthread.a
  EXTCAP_PATH = $(APPDATA)/Wireshark/extcap/
endif

all: $(BIN)

$(BIN): $(SRCS) $(HDRS)
	gcc $(CFLAGS) $(SRCS) -o $(BIN) $(LDFLAGS)

clean:
	rm -fr $(BIN)

install: $(BIN)
	cp $(BIN) $(EXTCAP_PATH)

prog_eeprom:
	./usb_sniffer --mcu-eeprom ../firmware/usb_sniffer.bin

prog_sram:
	./usb_sniffer --fpga-sram ../fpga/impl/usb_sniffer_impl.bit

prog_flash:
	./usb_sniffer --fpga-flash ../fpga/impl/usb_sniffer_impl.jed
```

ataradov, 2023-11-22:
I'll try to merge this into the main repo. I won't be able to test it, but at hopefully it will be helpful to some extent.

Does  -fno-diagnostics-show-caret not work on macOS?

fizzyade, 2023-11-23:
The macOS SDK is based on clang, there’s some sort of shim that presents it as gcc when you call gcc, but it doesn’t support whatever that switch does.

There’s also some issue with the latest Wireshark, if I install it then the usb sniffer device doesn’t appear, if I reinstall the old version I had (4.0.5) the plugin reappears, I’m guessing they’ve changed something under the hood or something, I need to investigate.

Adrian

ataradov, 2023-11-23:
I've updated the Makefile. That option is cosmetic, it disables spam on errors, which I hate. And since development is done on Linux, I just moved it under the Linux part.

fizzyade, 2023-11-23:
> I've updated the Makefile. That option is cosmetic, it disables spam on errors, which I hate. And since development is done on Linux, I just moved it under the Linux part.

Perfect, just pulled it to double check and it's working, thanks for accepting the changes.  

I need to have a look at why it's not working in the latest Wireshark on macOS, I don't know if this is a generic problem or something related just to macOS (I'm going to assume it's the latter), I will report back, but the build here works in 4.0.5 on my x86 Mac.

I guess it's also obvious enough if it fails to build that it requires brew (since that's the easiest way of accessing ports of "linux" based tools/libraries under macOS).   If somebody has found themselves here then they're probably a developer.

but just Incase somebody searches the issues...

if they don't have Xcode installed, then they can either download it via the App Store, if you don't want to take up 30GB of space then you can install the command line tools only.

```bash
xcode-select --install
```

brew can then be installed with:

```bash
bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"
`

and then the 2 dependencies required build the sniffer software can then be installed with:

```bash
brew install pkg-config
brew install libusb
```

fizzyade, 2023-11-23:
on 4.0.5 (which is what I originally had installed and tried first before upgrading), it looks for the ext cap plugins in: 

`
~/.config/wireshark/extcap
`

on Wireshark 4.2.0 on macOS it looks for the personal ext cap plugins in:

`
~/.local/lib/wireshark/extcap
`

It's now showing up and working correctly when copied to this different location.

casandra9, 2024-01-07:
Thank you both! Works like a charm. I get pretty constant 49.1MHz on my iMac.
I confirm the wireshark extcap path to be in ~/.local/lib/wireshark/extcap

I have modified the export path in the Makefile accordingly before ‘make’.

billswartz7, 2025-06-13:
First I want to thank you for all of your work!   I now use this in my embedded systems class to show how USB works at the low level.   The open design process was especially useful.

Currently, we are using WireShark Version 4.4.7 (v4.4.7-0-g7980339b1630) native MacOSX and the proper extcap directory for the install of usb_sniffer executable is

/Applications/Wireshark.app/Contents/MacOS/extcap

This may be of use to anyone going down this path.    Thanks again!

closed 2026-10-10 by ataradov