ataradov/usb-sniffer #23: Issue with high speed devices

aozima, 2023-09-18:
Hi,

I've built 5pcs boards, works fine with full-speed devices. 

I try to connect any high-speed device, e.g: Udisk or Card Reader。

The PC can enumeration  and  read/write disk, but Wireshark can't capture data.

test hardware

```
$ sudo ./usb_sniffer_linux --test
Starting speed test
Transfer rate: 48.05 MB/s
Transfer rate: 48.26 MB/s
Transfer rate: 48.25 MB/s
Transfer rate: 48.12 MB/s
```

aozima, 2023-09-18:
case 1:
only `Detected speed: Full-Speed`

```
Protocol	Length	 	Info
Syslog	27		USB PHY error
USBLL	1		Invalid Packet ID (0xff)
Syslog	39		Line state: J (583.00 ns)
Syslog	40		Detected speed: Full-Speed
Syslog	41		Line state: SE0 (257.70 us)
Syslog	46		Line state: K [single] (5.08 ms)
Syslog	40		Line state: SE0 (25.35 us)
Syslog	47		Line state: K [single] (49.00 us)
Syslog	47		Line state: J [single] (49.00 us)
```

ataradov, 2023-09-18:
Did you configure the high-speed mode in the interface settings? The detection message is just information, it does not actually change the capture speed.

closed 2023-09-18 by aozima