ataradov/usb-sniffer #24: LCMXO2-2000HC-5TG100C

imayka, 2023-10-15:
Hello. Can I replace LCMXO2-2000HC-5TG100C with LCMXO2-2000HC-5TG100I ? LCSC has both of them but prices are 65$ vs 15$ correspondingly. Should I modify or recompile FPGA project if yes? Thank you!

imayka, 2023-10-15:
Ok, so the difference between Industrial or Commercial is only Operating Temperature. But will firmware prepared for C device work with I device without modification (recompiling)?

ataradov, 2023-10-15:
You can use any temperature variant without changing the binary.

ataradov, 2023-10-15:
For me it shows both out of stock, but industrial is available pre-order, so it has the real updated price. Commercial one still has outrageous shortage price listed.

imayka, 2023-10-15:
> You can use any temperature variant without changing the binary.

Thank you!

closed 2023-10-15 by imayka