ataradov/usb-sniffer #25: The on-board LEDS

fizzyade, 2023-11-22:
It would be useful to have through hole pad options for the on board LEDS, once the PCB is in an enclosure the LED's are no longer visible, I can obviously modify the hardware to provide these PTH holes, it's a 2 minute job, but I figure that it might be a useful thing for other people since an enclosure protects the PCB, so I figured I'd ask here to see if threres any interest in providing these pads in the origin repo.  I could solder coil wire directly to the SMD pads, but my concern there would be that it would be relatively easy to rip the pad off during later disassembly if I forget about it.

The pictures here show my case design, I could put the leds on a surface on the main body or on the end caps.

Many thanks for the brilliant project!

[image]
[image]

ataradov, 2023-11-22:
For this design, I don't really want to do anything with the LEDs. In my case I relied on the gray plastic being transparent, and it works well enough. LEDs here are not all that useful anyway, and I don't want to create a hardware revision just for that.

I hate working with through hole LEDs, so I avoid them in my designs, but since I'm also making 3D printed cases, it does become an issue. And I'm trying to figure out how to address this. But my experiments are focused on making clam shell cases and printing in holes for the light pipes or something like this.

fizzyade, 2023-11-23:
No problem, I figured that since they weren’t there then that would be the answer!  I’ll just modify it myself (thanks for the open source design!)

It wasn’t so much about actually mounting PTH LED’s directly onto the board, it was purely just providing a PTH hole so that the location of the LED's don’t need to be tied to a specific location on the PCB, with a test point the wires could just as easily connect to a daughter board that has a bunch of SMD LED’s arranged is a coherent order.

It’s just a sanity check for me since my USB cables often work themselves loose on my laptop, so looking at the LED’s is an easy confirmation that at least power is getting to the device from the host.

Adrian

KarlK90, 2024-02-18:
@fizzyade Hi Adrian, I like your take on the case very much, would you mind sharing the case files?

fizzyade, 2024-02-18:
Yeah no problem, I will upload them later to printables, remind me tomorrow if I haven't replied with the link.

fizzyade, 2024-02-18:
> @fizzyade Hi Adrian, I like your take on the case very much. ~would you mind sharing the case files~? Already found it here https://www.printables.com/model/657228 - thanks for sharing.

Yeah, I couldn't log in earlier and when I just managed to get in I realised I'd already shared it!

closed 2026-10-10 by ataradov