ataradov/usb-sniffer #31: Help flashing to eeprom

anujdeshpande, 2024-01-31:
Hey @ataradov - I am trying to assemble this, but running into issues with the FPGA IDCODE. Is this something you have seen before? Any tips on what could be the issue?

```
$ ./usb_sniffer_linux --mcu-sram usb_sniffer.bin
Uploading 3187 bytes into the FX2LP SRAM
...done
$ ./usb_sniffer_linux  --mcu-eeprom usb_sniffer.bin
Error: incorrect FPGA IDCODE (0x00000000)
$
```
On kernel.log, I see

```
Jan 31 13:04:14 mbradmin-System-Product-Name kernel: [938797.213555] usb 1-6: new high-speed USB device number 25 using xhci_hcd
Jan 31 13:04:15 mbradmin-System-Product-Name kernel: [938797.361730] usb 1-6: New USB device found, idVendor=04b4, idProduct=8613, bcdDevice=a0.01
Jan 31 13:04:15 mbradmin-System-Product-Name kernel: [938797.361733] usb 1-6: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Jan 31 13:04:15 mbradmin-System-Product-Name kernel: [938797.362149] usbtest 1-6:1.0: FX2 device
Jan 31 13:04:15 mbradmin-System-Product-Name kernel: [938797.362152] usbtest 1-6:1.0: high-speed {control bulk-in bulk-out} tests (+alt)
Jan 31 13:04:57 mbradmin-System-Product-Name kernel: [938840.009639] usb 1-6: USB disconnect, device number 25
Jan 31 13:04:58 mbradmin-System-Product-Name kernel: [938840.529418] usb 1-6: new high-speed USB device number 26 using xhci_hcd
Jan 31 13:04:58 mbradmin-System-Product-Name kernel: [938840.680478] usb 1-6: New USB device found, idVendor=6666, idProduct=6620, bcdDevice= 1.00
Jan 31 13:04:58 mbradmin-System-Product-Name kernel: [938840.680481] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 31 13:04:58 mbradmin-System-Product-Name kernel: [938840.680482] usb 1-6: Product: USB Sniffer
Jan 31 13:04:58 mbradmin-System-Product-Name kernel: [938840.680483] usb 1-6: Manufacturer: Alex Taradov
Jan 31 13:04:58 mbradmin-System-Product-Name kernel: [938840.680484] usb 1-6: SerialNumber: [-----SN-----]
```

ataradov, 2024-01-31:
Double check that your FPGA is soldered correctly and that it is a correct device. Post pictures of the assembled board. There is not much you can tell from the logs, since you will get the same exact result even if FPGA is not soldered at all.

anujdeshpande, 2024-02-01:
@ataradov here's a pic -

[image]

ataradov, 2024-02-01:
Your FPGA is soldered in the wrong orientation. Pin 1 is marked by the small dot.

anujdeshpande, 2024-02-01:
Yeah - just figured it out. I feel stupid now :/
Thanks for the help

closed 2024-02-01 by anujdeshpande