ataradov/usb-sniffer #32: Improvement: please move GND and 3V3 vias from FPGA pins

acassis, 2024-02-09:
Hi Alex, first of all thank you for this great project.

I had a hard time doing hardware debug because of these vias near FPGA pins:

[image]

Because I'm soldering manually with hot iron, the solder move very easily to these vias. I removed many components until I found this issue. After cutting the wire from C8 to 3V3 in the other side of the board the issue disappeared.

Maybe you could move the vias from C5, C7 and C9 too, because they also could end up causing similar issue.

ataradov, 2024-02-09:
I won't be making hardware changes just for that. And I don't expect any other hardware changes. This is my standard clearance and I never had any issues soldering those devices by hand. 

Provided Gerber files specify tented vias. You would have to try really hard to create shorts. I don't even know how I would do that if I wanted to.

You likely need to use a better soldering equipment or techniques.

acassis, 2024-02-09:
Hi @ataradov,
actually I just soldered using the standard way I used to solder many other projects (that is basically as [doogulass](https://www.youtube.com/@doogulass) did in this video as well.

So no, I didn't "tried hard to create the shorts".

I only opened it issue to help making it easier for other people doing it, I didn't expect you receiving it as a criticism. The only purpose was to avoid other people facing same issues I faced. I manufactured the board at JLCPCB that has a good quality, so it was not the soldermask getting removed easily.

I suspect the solder flowed from bottom side (lower pad of C8) and made contact with some FPGA pin or the GND via at side of it. BTW, this is just a supposition because I don't have x-ray machine here to confirm my suspect.

ataradov, 2024-02-09:
Your description does not match the possible short you are describing.  If the short is under the IC, how would removing C8 help it? It is much more likely that C8 itself has a short.

The solder flow you are describing is extremely hard to achieve.

acassis, 2024-02-10:
I didn't remove C8, I just did a sharp cut the wire trace from C8 via to 3V3 (that I cold re-wire again with a drop of solder) and the short disappeared.

ataradov, 2024-02-10:
Which is equivalent to removing the C8 from the circuit. If the short was on the other side of the board under the IC, then doing that cut would not have solved the issue.

If you still have C8 disconnected, then you can measure it for continuity. If there was a short, it would be in C8.

closed 2024-03-18 by acassis