ataradov/usb-sniffer #33: Transfer rate:  0.00 MB/s

Spoosie, 2024-03-08:
Hi, and thanks for this cool project. I have build the hardware and programmed the MCU und FPGA. Everything worked fine until the test procedure. 
I got 0.00 MB/s at the transfer speed, and i have no clue what i can do to fix that! 
Could you give me a hint?

ataradov, 2024-03-08:
Most likely you have some issues with the USB PHY IC and it is not producing 60 MHz clock. Up to the test point it would matter, but the test part needs it.

So, measure that clock and the input 26 MHz clock.

acassis, 2024-03-19:
I faced similar issue: https://www.youtube.com/watch?v=3Boyl4IId2g

I checked the 26MHz clock and it is fine, so probably is some issue on USB3343, right?

ataradov, 2024-03-19:
Do you see 60 MHz as well? If not, then yes, something is wrong with the USB PHY. It should start producing 60 MHz clock for the FPGA. Without it programming will work, but no data will be sent.

ataradov, 2024-03-19:
And the most obvious thing to check that is it soldered in a correct orientation.

acassis, 2024-03-19:
I think the orientation is correct because I looked the datasheet before soldering, see attached image.

The 60MHz should be on pin 2 of USB3343, correct? So, that could be the issue, there is nothing on this pin!
[image]

ataradov, 2024-03-19:
Yes, this looks like a correct orientation. And yes, pin 2 should have 60 MHz.

Another thing to double check is that you have  8.06K (R11) resistor mounted and the value is correct. Apparently, it does not produce any clock when this resistor is not mounted.

acassis, 2024-03-19:
Yes, it is mounted, but I checked on multimeter and its value is 8.25K. Is it inside the tolerance range?

ataradov, 2024-03-19:
It needs to be 8.06 +/- 1%. So, it should be in the range 7.98 - 8.14 kOhm. I don't know what will happen if it is outside of that range. I don't think it is that sensitive for the purpose of the clock generation, but who knows.

jpenalbae, 2024-05-14:
I have had this same issue and wanted to share here in case others happen to find this thread when searching on how to solve the problem.

In mi case it was bad soldering on the USB3343. I guess it can be a bit tricky to solder by hand...

I fixed it just by de-soldering and re-soldering the USB3343. First remove all the solder in the IC and the board. Prime the ground pad with very little solder, using the heat gun solder the IC as usual, press down with a pair of tweezers on the IC until everything cools down to make sure it properly seats and its not floating on the solder pad. Next using a good amount of flux solder the external pins, I would recommend drag soldering technique there.

acassis, 2024-05-14:
@jpenalbae Thank you Jaime, that was my suspicious too! I didn't have the change to resolder it yet. How are you testing it? Did you connect some device to J2 (USB-C) connector during the speed test?

ataradov, 2024-05-14:
Speed test only needs host connection, target can be disconnected. The only reason USB3343 is necessary for the test to work is that it provides 60 MHz clock that is needed for everything to work.

jpenalbae, 2024-05-14:
> @jpenalbae Thank you Jaime, that was my suspicious too! I didn't have the change to resolder it yet. How are you testing it? Did you connect some device to J2 (USB-C) connector during the speed test?

I would try to re-soldering the same USB3343 you have in the board right now, that is what I did. If that didn't work I was planing on trying with a new one, but it wasn't necessary in my case.

For testing, as have been already mentioned around here, make sure 26MHz clock in, 60MHz clock out, download the datasheet for USB3343 and double check everything is in place. For my case it was pretty obvious cause besides the 60MHz clock, the integrated 1.8v LDO (pin 23) wasn't working either, so it was a bad USB3343 or a bad solder job.

kevinaangstadt, 2024-06-07:
I have the same issue, as well; however, I am able to measure 60MHz at the FPGA.  Any pointers on where to look next?

ataradov, 2024-06-07:
Was FPGA firmware upload correctly? Is the FPGA type (including speed grade) correct?

kevinaangstadt, 2024-06-07:
> Was FPGA firmware upload correctly? Is the FPGA type (including speed grade) correct?

Flashing the FPGA appears to succeed:

```shell
$ ./usb_sniffer_linux --fpga-flash usb_sniffer_impl.jed 
Programming FPGA flash
Erasing flash
Programming configuration data .....
Verifying configuration data
Programming and verifying Feature Row
Programming and verifying FEABITS
Exiting programming mode
...done
```

I purchased a LCMXO2-2000HC-5TG100C from DigiKey, which I believe matches what is specified in the BOM.

ataradov, 2024-06-07:
Yes, this all this is correct. The other thing to check is that you have 30 MHz on the IFCLK (FPGA pin 88). This clock is generated by the MCU and FPGA is using it to send the data.

And check CTRL_CLK/CTRL_DATA. Those pins are used to enable the test mode. There is no persistent clock on them, but they should some activity when the test mode is started. 

All those signals go from the MCU to the FPGA, so check on the FPGA side, so you verify soldering on both sides at the same time.

kevinaangstadt, 2024-06-07:
Thanks! I found a few loose pins on the MCU. After fixing those, I'm seeing 40-50MB/s as expected.

Spoosie, 2024-06-24:
> Hi, and thanks for this cool project. I have build the hardware and programmed the MCU und FPGA. Everything worked fine until the test procedure. I got 0.00 MB/s at the transfer speed, and i have no clue what i can do to fix that! Could you give me a hint?

Okay, after a long time i came back to this project. And i found a heavy bug. I had to rotate the USB3343 about 90 degrees. Uhaa.. hopefully the chip was not damaged! 

But after a short test i got the desired 40-50MB/s. That's fine!

Now i got some other errors with the wireshark capture. I would like to test some sticks laying around, but all of them are not really send the right datas. I got "USB PHY Error". Could it be, that the USB3343 is faulty because of my lack of competence? :-/ Maybe i should order a new one, and try it out.

ataradov, 2024-06-24:
Did you configure the correct capture speed? If yes, then I would assume the PHY is damaged and replace it before spending more time investigating that.

closed 2026-10-10 by ataradov