ataradov/usb-sniffer #34: USB failing during the enumeration

acassis, 2024-03-18:
Hi Alex, because that short-circuit issue under the FPGA I decided to take a different approach: starts from basic and get USB Host working!

I soldered only the USB Host (all the components of Kicad schematic page 3, the J1 connector and its resistors and IC2 regulators and its components), after that I tried to connect it to computer it was detected and almost immediately it was disconnected:

```
[35566.856268] usb 3-5: new high-speed USB device number 5 using xhci_hcd
[35567.072707] usb 3-5: New USB device found, idVendor=04b4, idProduct=8613, bcdDevice=a0.01
[35567.072721] usb 3-5: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[35567.224368] usbtest 3-5:1.0: FX2 device
[35567.224373] usbtest 3-5:1.0: high-speed {control bulk-in bulk-out} tests (+alt)
[35567.224434] usbcore: registered new interface driver usbtest
[35571.954145] usb 3-5: USB disconnect, device number 5
[35572.328136] usb 3-5: new high-speed USB device number 6 using xhci_hcd
[35572.544706] usb 3-5: New USB device found, idVendor=04b4, idProduct=8613, bcdDevice=a0.01
[35572.544718] usb 3-5: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[35572.545598] usbtest 3-5:1.0: FX2 device
[35572.545605] usbtest 3-5:1.0: high-speed {control bulk-in bulk-out} tests (+alt)
[35575.855084] usb 3-5: USB disconnect, device number 6

$ sudo lsusb 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 003: ID 0c45:672e Microdia Integrated_Webcam_HD
Bus 003 Device 002: ID 27c6:63ac Shenzhen Goodix Technology Co.,Ltd. Goodix USB2.0 MISC
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
$
```

Initially I was thinking my 24MHz oscillator was faulty, because I'm using the ECS-3225MVLC-240-CN-TR that @casandra9 suggested, but I looked at CKOUT pin in the oscilloscope and it is generating 12MHz as expected.

Did you face similar issue in the past? How to debug/fix this issue?

ataradov, 2024-03-18:
This looks like a poor physical connection in the connector or a broken USB cable or something like that. If it gets to that point, it means it is mostly working and clocks are fine.

I would do this - connect the cable on the PCB side and make sure that PCB plus that end of the cable is not moving. Then plug in the other end of the cable into the PC. See if the behavior repeats. With a poor connection and a stationary PCB I would expect it to either work or not, but not this intermittent behavior.

casandra9, 2024-03-19:
[Quote](…) New USB device strings: Mfr=0, Product=0, SerialNumber=0 (…)[\Quote]

Could it be that you didn’t program the FW?

ataradov, 2024-03-19:
This is just a raw blank FX2LP device. This enumeration happens on the hardware level, there is no firmware at this point. But it still should not disconnect. There is no real point in programming the real FW if this mode does not work reliably. The real FW won't either.

acassis, 2024-03-19:
Thank you @ataradov I'll verify the USB-C soldering, because I didn't use much solder on its pads to avoid short-circuiting the pads, maybe it is not well soldered.

@casandra9 as @ataradov explained it is a blank device, so it is a chicken-egg dilemma, because the device is disconnecting just after plugging the case I cannot flash the firmware yet.

acassis, 2024-03-19:
In fact it was the USB-C connector that wasn't well soldered:

```
[ 1844.234868] usb 3-3: new high-speed USB device number 5 using xhci_hcd
[ 1844.383410] usb 3-3: New USB device found, idVendor=04b4, idProduct=8613, bcdDevice=a0.01
[ 1844.383414] usb 3-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1844.511175] usbtest 3-3:1.0: FX2 device
[ 1844.511178] usbtest 3-3:1.0: high-speed {control bulk-in bulk-out} tests (+alt)
[ 1844.511228] usbcore: registered new interface driver usbtest
[ 1850.322958] ucsi_acpi USBC000:00: GET_CONNECTOR_STATUS failed (-110)
[ 1856.466951] ucsi_acpi USBC000:00: GET_CONNECTOR_STATUS failed (-110)
alan@dev:~$ sudo lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 04b4:8613 Cypress Semiconductor Corp. CY7C68013 EZ-USB FX2 USB 2.0 Development Kit
Bus 003 Device 004: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 003: ID 0c45:672e Microdia Integrated_Webcam_HD
Bus 003 Device 002: ID 27c6:63ac Shenzhen Goodix Technology Co.,Ltd. Goodix USB2.0 MISC
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
$ 
```

Thank you Alex! Let's move on!

closed 2024-03-19 by acassis