ataradov/usb-sniffer #35: usb_sniffer_linux failed with latest linux wireshark 4.2.5

mosquito520, 2024-05-16:
Hi

I've using linux mint 21.2 (based on Ubuntu Jammy)
When I install system package wireshark 3.6.2.2
I can select USB sniffer on wireshark list interface, and I can receive data
not able to decode, but I can find different package size during enumeration

Due to Wireshark 3.x does not support usb decode
so I've using PPA to install latest wireshark 4.2.5
when I upgrade Wireshark to 4.2.5, I'm not able to select USB sniffer on wireshark anymore.

Does usb_sniffer_linux test with 4.2.5 before?

Thanks

ataradov, 2024-05-16:
Note that you may need to create this directory if it already does not exists.

mosquito520, 2024-05-16:
> At some point Wireshark updated the location for the extcap plugins. The resent versions seem to use "~/.local/lib/wireshark/extcap". I have updated the Makefile accordingly.
> 
> In order to find out what this path is on your system, look in Help -> About Wireshark -> Folders -> Personal Extcap path.

Thanks for reply, I'll verify it tomorrow:)

mosquito520, 2024-05-17:
Hi Alex
Thanks for reply !
I've confirm it work when change to "~/.local/lib/wireshark/extcap"
Also
> In order to find out what this path is on your system, look in Help -> About Wireshark -> Folders -> Personal Extcap path."
is clear !

But I'll suggest https://github.com/ataradov/usb-sniffer?tab=readme-ov-file#installation
Maybe update this description would be nice

Thanks for your effort! :smile:

closed 2024-05-17 by mosquito520

ataradov, 2024-05-17:
Fixed the readme.