ataradov/usb-sniffer #36: Digikey BOM & Comparison to Lite

dderiso, 2024-05-30:
Dear Alex,

Thank you for sharing such a great project!

If it helps others, here is a DigiKey cart: https://www.digikey.com/short/jqrb2558

Do these parts look about right? Do you happen to know your USB MPNs?

Cheers,
Dave

PS. 

This RPi version also looks great

https://github.com/ataradov/usb-sniffer-lite

Which do you recommend for new users?

ataradov, 2024-05-30:
The list seems fine, but the prices compared to LCSC are steep.

USB connectors used here are very generic, there are dozens of options that would work. On a quick look, this should work:
https://www.digikey.com/en/products/detail/amphenol-cs-commercial-products/12402012E212A/13683192
https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/6364372-2/2286712

The USB-A side is designed for a connector with flat mounting pins, but if you are doing manual assembly, then more common connectors like this https://www.digikey.com/en/products/detail/on-shore-technology-inc/USB-A1HSB6/2677751 would also work, you may just need to straighten out the mounting pins.

If you have a way to build the full one, I would always recommend doing that. The Lite one is good because most people will have a Pi Pico and an old USB cable, so anyone can build it in 15 minutes.

dderiso, 2024-05-31:
Awesome! Thank you!