ataradov/usb-sniffer #40: Upstream LCMXO2-2000HC-5TG100

phodina, 2024-07-06:
Hi @ataradov ,
thanks for great project to analyze USB traffic. I notice the LCMXO2-2000HC-5TG100 component is not part of Kicad libaries.

I've modified the component and split it into 2 units. Maybe it would be better to split it into more based on the io banks. Any suggestions?

https://gitlab.com/kicad/libraries/kicad-symbols/-/merge_requests/4600

ataradov, 2024-07-06:
I don't use any of the standard components. I don't like to depend on the outside sources, since I can't be sure of the quality and consistency.

Whether things should be divided - I personally don't like it. I think it is far easier to keep track of things when there is a single block representing the whole device. It obviously becomes impractical for very high pin count devices, but for 100 pins it works fine. And this personal preference is another reason why I only use my own symbols.

phodina, 2024-07-07:
Well, the point in upstreaming the components is to have them comply with certain rules. Check the [KLC](https://klc.kicad.org/) and only if they follow them they will be part of the library and ensures quality.

I agree that if the pin count is low there's no point. However, for larger pincount it's better as it does not overflow or take most of the sheet space and can be logically divided.

I just ask if you have suggestion as I hesitate the break it apart since it's 100 pins and that's still manageable in single symbol.

ataradov, 2024-07-07:
Sure, but I personally have no desire to upstream anything. And you feel free to do whatever is needed. 

I disagree with KiCad's (or any other CAD for that matter) component rules, so I don't use them in my projects. I'm perfectly fine maintaining my own library.

closed 2026-10-10 by ataradov