ataradov/usb-sniffer #41: Smaller packages - QFN or BGA

phodina, 2024-07-06:
Hi @ataradov,

could you please suggest components that take less space and provide the same functionality? I mean are there compatible parts Cypress CY7C68013A MCU, Lattice LCMXO2 FPGA in QFN or BGA packages?

I understand that these part will impact the necessary skills to solder the components at home.

However, it would be nice if it could be fitted into smaller form factor like Framework module.
https://github.com/FrameworkComputer/ExpansionCards/tree/main/Mechanical

ataradov, 2024-07-06:
CY7C68013A is available in QFN-56 and it should work exactly the same. 
XO2-2000 is available in 132 pin BGA package. I don't know if the same pins are available on this package, FPGA project changes would likely be required.

At the same time, physical dimension are defined by the target connectors. Their layout is fixed and limited by the cable connectors. You can't place them much closer or make USB-A connector smaller. 

Plus I don't see what is the benefit of making that a module. The sniffer is meant to be an external tool.

phodina, 2024-07-07:
Thanks for the components. Yes, probably there would be need to generate new bitstream for the FPGA.

Sure, the connectors can be downscaled and in theory it does not matter as you say it should be used externally.

closed 2024-07-07 by phodina

Basilisvirus, 2024-12-26:
Note that the existing CY7C68013A-56PVXC IC has been marked as end of life.  
[image]

ataradov, 2024-12-26:
I would expect them to be available for a long time and there are at least 3 different clones. All the new stuff from Cypress is in unusable packages, so I'll pass on moving on to that.