ataradov/usb-sniffer #45: EEPROM programing failes with 'Error: usb_i2c_read(): LIBUSB_ERROR_TIMEOUT'

Morpheus-79, 2024-09-09:
I can't finish the build since i'm stuck at program the EEPROM. Loading the firmware into the SRAM works fine and:

_usb_sniffer_win.exe --mcu-sram usb_sniffer.bin_

... outputs:

_Uploading 3187 bytes into the FX2LP SRAM_
_...done_

But when trying:

_usb_sniffer_win.exe --mcu-eeprom usb_sniffer.bin_

... i always get:

_Programming 3187 bytes into the FX2LP EEPROM (SN: 44382301285c08)_
_Error: usb_i2c_read(): LIBUSB_ERROR_TIMEOUT_

I've tried different USB ports (connected to different USB host controllers) and tried programing the EEPROM under Linux too -> but every attempt ends with:

_Error: usb_i2c_read(): LIBUSB_ERROR_TIMEOUT_

I opted for some alternate parts, since the ones from the BOM weren't available at the time. For example i got a HG24LC128M/TR EEPROM:

https://www.lcsc.com/product-detail/EEPROM_HGSEMI-HG24LC128M-TR_C3007884.html

... and some different crystal oscillators:

https://www.lcsc.com/product-detail/Crystal-Oscillators_Abracon-LLC-ASEDV-26-000MHZ-LR-T_C1857762.html
https://www.lcsc.com/product-detail/Crystal-Oscillators_Abracon-LLC-ASEDV-24-000MHZ-LR-T_C17564399.html

But this shouldn't be a problem... at least: i thought. Or did i miss something and one of these components is the culprit?

ataradov, 2024-09-09:
The problem would be somewhere around EEPROM. The oscillators are ok if it gets that far.

But this error is strange. This can only happen during data verification, which means that EEPROM worked well enough to at least pretend to write the data.

Build the PC application from code and put debug prints in the loop in the fx2lp_eeprom_upload() to see what iteration fails. This is the only place that does the reads.

ataradov, 2024-09-09:
Also, EEPROM may be extremely slow. So, I would also try to place os_sleep(10); between fx2lp_eeprom_write() and fx2lp_eeprom_read() in the same function.

Morpheus-79, 2024-09-09:
I've added some debug prints and got a read error at the start address after a successful write:

_Programming 3187 bytes into the FX2LP EEPROM (SN: 44382301285c08)_
_Writing to EEPROM at address 0x0_
_Write complete._
_Reading from EEPROM at address 0x0_
_Error: usb_i2c_read(): LIBUSB_ERROR_TIMEOUT_

Trying to write to the EEPROM a second time results in a write error at the start address:

_Programming 3187 bytes into the FX2LP EEPROM (SN: 44382301285c08)_
_Writing to EEPROM at address 0x0_
_Error: usb_i2c_write(): LIBUSB_ERROR_PIPE_

Since i've had a MiniPro TL866 laying around somewhere in the chaos of my home, i desoldered the EEPROM and tried to read it. After selecting an 24LC128 as chip device i was able to read the EEPROM... but it contained only a couple of the first bytes of the BIN file.

I resoldered the EEPROM. And since every writing attempt results in a corrupt firmware: i had to connect the board with the BOOT and VCC test points shorted to be able to give it another try. Loading the firmware into the SRAM worked as always. This time i tried to write to the EEPROM with a added _os_sleep(10)_ between writing and reading... but without any success:

_Programming 3187 bytes into the FX2LP EEPROM (SN: 44382301285c08)_
_Writing to EEPROM at address 0x0_
_Write complete. Sleeping..._
_Reading from EEPROM at address 0x0_
_Error: usb_i2c_read(): LIBUSB_ERROR_TIMEOUT_

I've now ordered some genuine Microchip 24LC32A EEPROMs from a local seller. Hopefully it'll fix this strange behaviour.

ataradov, 2024-09-09:
Yeah, I would not spend too much time debugging that. I would assume that something is fishy with those EEPROMs.

ataradov, 2024-09-09:
Also, you should be able to use the sniffer, you just need to upload the firmware every time you power cycle it. It will not have the correct serial number, but this should not cause any issues.

Morpheus-79, 2024-09-11:
You were absolutely right:

The 24LC32A EEPROMs arrived today... i replaced the fishy one... and:

_Programming 3187 bytes into the FX2LP EEPROM (SN: 44382301285c08)_
_...done_

... the problem was fixed.

Thanks for taking the time to help... and for this awesome project!

closed 2024-09-11 by Morpheus-79