ataradov/usb-sniffer #47: InValid Packet ID ,When plugging in the USB cable to be tested,

blueoce, 2024-11-19:
1 the following figure,The correct way to plug in the USB cable to be tested is as follows:
[image]

2 the following figure,The correct display when plugging in the USB cable to be tested is as follows:
[image]

issue?
Does the 3343 chip have an issue? However, it works fine in other situations; the problem only occurs when plugging or unplugging the USB cable.

ataradov, 2024-11-19:
It is normal to see some connection noise (bouncing) when making a mechanical connection. The sniffer interprets that noise as data.

This is why all USB host stacks let the signals settle for 100-200 ms and then issue a bus reset when a new device is connected.

blueoce, 2024-11-19:
Can the error frames in Figure 2 be hidden, like in Figure 1，"If I encounter this issue, the only solution is to replace the USB 3343 chip！

blueoce, 2024-11-19:
Although replacing the USB3343 chip solved the problem, it suggests that the USB3343 was faulty. However, other functions were working normally. Is the USB3343 really faulty? If it is not faulty, can it be optimized to not display error frames? If the USB3343 is indeed faulty, what commands can be used to test and confirm that it is faulty?

ataradov, 2024-11-19:
Why do you want to hide them? They are representation of what happens on the bus. If you probe it with an oscilloscope you will see that bouncing too.

blueoce, 2024-11-19:
I disagree with the statement that replacing the chip is not helpful. It really did work after the replacement, I'm not lying to you。It is possible that some PHYs have slight manufacturing differences and are less sensitive to the noise.I think this is the real reason.

ataradov, 2024-11-19:
Well, you can continue to replace them if you want. But if anything, you are getting a worse deal. I personally would like to observe the real bus behavior. 

I feel like you are missing the point of this tool or need something else for your use case.

stappersg, 2024-11-19:
On Tue, Nov 19, 2024 at 09:00:47AM -0800, Alex Taradov wrote:
> 
> I feel like you are missing the point of this tool or need something else for your use case.
> 

Act upon feeling and close this issue.

(Do known that closed issues can updated with information
as "Why do you want to hide the noise that does exist?")

Groeten
Geert Stappers
-- 
Silence is hard to parse

closed 2024-11-19 by ataradov