ataradov/usb-sniffer #50: EP0 (Control) - Set Feature/ Get Feature capture support

MohamedHassanNasr, 2024-12-21:
Hi,
First of all thanks so much for this great project, finally we have a good working USB sniffer for less than 50$, i have a project for simulating PS5 DualSense controller and i want to capture some traffics between console and controller using usb sniffer, and i had successfully captured traffic between console and controller except when console send Set Feature Request (F0) which is special request that handled by EP0, and also Get Feature Requests (F1, F2) which handled by EP0 also, so is there is any way to support capturing this requests?

Best regards,
Abkarino

ataradov, 2024-12-21:
The sniffer captures raw data on the bus, it does not know what any of the requests are. The decoding happen on the host and done by Wireshark.

Do you not see the frames at all or just no decode?

Also, can you clarify what specific requests you are referring to? I don't recognize those IDs (?) as anything standard.

MohamedHassanNasr, 2024-12-21:
Thanks so much for your fast response,
below you can find HID report descriptor:

0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x15, 0x00,        // Logical Minimum (0)
0x26, 0xFF, 0x00,  // Logical Maximum (255)
0x75, 0x08,        // Report Size (8)
0x95, 0x3F,        // Report Count (63)
0x09, 0x40,        // Usage (0x40)
0xA1, 0x01,        // Collection (Application)
0x85, 0xF0,        //   Report ID (0xF0)
0x09, 0x47,        //   Usage (0x47)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x85, 0xF1,        //   Report ID (0xF1)
0x09, 0x48,        //   Usage (0x48)
0x95, 0x3F,        //   Report Count (63)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x85, 0xF2,        //   Report ID (0xF2)
0x09, 0x49,        //   Usage (0x49)
0x95, 0x0F,        //   Report Count (15)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 48 bytes

i tried to look at what wireshark show, but i did not find this request yet in wireshark logs.

ataradov, 2024-12-21:
Those are not going to be send as Feature request. They will be Get and Set Reports commands (HID) and they will be sent to the endpoints of the HID device, not Control endpoint.

MohamedHassanNasr, 2024-12-21:
Thanks for clarification, I’m not so familiar with USB requests, anyway what is the reason if I can not find this requests in my logs?
By the way the DS controller is HS device.

ataradov, 2024-12-21:
Those reports will be sent in response to user inputs on the controller. Start the capture and move the joysticks around and press the buttons and you should see corresponding reports sent by the controller.

Also, there seems to be a whole bunch of info on that controller already including this tool https://nondebug.github.io/dualsense/dualsense-explorer.html that uses WebHID to interact with it. You can try to use it to further generate traffic between the controller and the host.

MohamedHassanNasr, 2024-12-21:
Thanks for updates, actually this requests F0, F1, F2 are sent by host (PlayStation 5) to controller once user press PS button to link controller to host, and I already did that and pressed many keys to capture normal input/output requests with no luck at all.

ataradov, 2024-12-21:
Well, I have no idea what they are doing there specifically. HID is very flexible, so all that vendor specific stuff can be whatever. But no matter the direction, they will be accessed by HID Get and Set Report commands using that ID (F0, F1, F2) as a report ID. They are not commands, they are just IDs uses as one of the fields in the command.

In any case, if you see other frames, but not those commands, then they are likely not there.

MohamedHassanNasr, 2024-12-21:
Ok, I’m sure it is there, because console can not accept input from controller until this requests are passed successfully, anyway someone I know had captured this traffic used beagle USB analyzer, I’ll share his dump if it may help to know why this traffic is not shown in my logs.

ataradov, 2024-12-21:
I'm not going to reverse engineer that stuff, I have no real interest in that. This is not a tool issue, it captures everything that is present on the bus. The frames are either there and you are not finding them in the log, or they are not really necessary. Or if they are related to the pairing, they may be sent once after the initial connection.

MohamedHassanNasr, 2024-12-22:
No problem, I will try command line mode first without Wireshark maybe I can find requests on the logs, then it will be a Wireshark parsing issue.

ataradov, 2024-12-22:
Command line mode still just save a pcapng file. And you still need Wireshark to look at it.
Are you getting that much traffic that you can't find data in it? Do you have folding of the empty frames enabled? There should not be a lot of actual traffic for the HID device.

MohamedHassanNasr, 2024-12-22:
Understand, anyway thanks so much for your response, and for your very nice project and thanks also for open sourcing it, now you can consider this issue as closed.

stappersg, 2024-12-22:
On Sun, Dec 22, 2024 at 10:40:18AM -0800, Mohamed Hassan wrote:
> ... now you can consider this issue as closed.

Please click on the "close issue" button yourself.

Groeten
Geert Stappers
-- 
Silence is hard to parse

closed 2024-12-22 by ataradov