ataradov/usb-sniffer #51: Suggestions on BOM, PCB layout

Basilisvirus, 2024-12-26:
I will replicate this open source project soon. I am looking at the files right now and I would like a feedback from the suggestions Im making :) 

1. For easier hand-soldering, without the need of a hot air station I suggest IC4 to have a large via hole in the middle of it:  
[image]  

This way the middle pad can be soldered from the bottom side by filling it with solder.   
Be careful: make sure you slowly pour solder in the hole (start pouring solder to the side of the hole and wait for it get hot enough so that the solder will slow in on its own) and that the solder has "spourted" inside the hole on its own, to make sure the solder actually touched the pad and it didnt just cover the hole.
 
2. For mouser.com, IC6 (DSC6101CI2A-026.0000) is obsolete:  
[image]

I think DSC6003MI2B-026.0000 is a good replacement. It is also smaller in size (VFLGA-4 footprint) , which means less load capacitance, easier to solder on the pre-existing pad without how air since the SMD pads now stick out.:  
[image]

3. Same case for DSC6101CI2A-024.0000:  
[image]
The DSC6001HI2B-024.0000 (VFLGA-4 footprint as well)  looks like a good replacement. Its smaller and the pads (again) stick out, so easier to solder with hand soldering iron.

4. This doesnt matter really, for the frequencies and length of the traces, but you will have impedance mismatch when the copper changes width, to the places where I point the arrows. A solution to that is to make the width not change so much, aka make zones around the pads:  
[image]  
This happens in some other places on the board.  

(I will add new suggestions as I check more stuff)

ataradov, 2024-12-26:
The hole is up to you. The IC is small and reasonably easy to solder by hand as it is.
None of the PCB stuff really matters, you can do it if it makes you sleep better.
And for the crystals, there is already multiple discussions with replacements from LCSD. The footprint universal, there are dozens of possible alternatives.

closed 2024-12-26 by Basilisvirus

stappersg, 2024-12-27:
On Thu, Dec 26, 2024 at 07:28:56AM -0800, Christianidis Vasileios wrote:
> Closed #51 as completed.

For the PCB-layout, Yes.

For the Bill Of Materials change, hope I to see a merge request.

Groeten
Geert Stappers
-- 
Silence is hard to parse

ataradov, 2024-12-27:
I'm not going to accept merge requests. I need this repo to  match what I have. Different people will have access to the different parts and it is impossible to accommodate everyone. Plus things change over time. If you are building this, substitutions like that should be easy anyway.

Basilisvirus, 2024-12-28:
Agree with ataradov. Since he is so active on the issues, if you need any clarification about the BOM, one might as well raise an issue for it. Changing BOM will mean that ataradov will need to re-solder stuff and change things. This takes time and effort. 

If anyone manages anything better they can make an issue, fork the project on their github and make the desired changes.