ataradov/usb-sniffer #52: .inf file invalid under windows 10

WillemOuwerkerk, 2024-12-28:
Ha Alex,

I can't get de .INF file installed under Windows 10.
I get de message "invalid security certificate" 
Can you give a hint on how to correct that?

Kind regards,
Willem Ouwerkerk

stappersg, 2024-12-28:
On Sat, Dec 28, 2024 at 03:07:57AM -0800, Willem_O wrote:
>  ... Windows 10  ...  message "invalid security certificate" 
> Can you give a hint on how to correct that?

Contact Microsoft, ask for blessing of the .inf

ataradov, 2024-12-28:
See if you have some bypass option. None of my stuff is signed because I'm sure as hell not going to give MS money to workaround crappy security of their OS.

Alternatively, you can try to use Zadig to generate drivers, I think it makes signed drivers, but I'm not sure.

Or use some other way to program the initial contents of the EEPROM. That INF file is only needed one time for the blank FX2LP, once the firmware is programmed, all the needed drivers are installed automatically..

WillemOuwerkerk, 2025-01-02:
Thanks Alex,

Thanks for the sniffer and best wishes for 2025.

Both Windows and Linux Mint do accept the sniffer now and both load a driver.
Under Windows i can select it from within Wireshark, but it only shows low level data, how can i change that?
Under Linux i can see the driver is loaded but Wireshark does not show it.
Any hints?

Kind regards,
Willem Ouwerkerk

ataradov, 2025-01-02:
What do you mean by "low data level"?
And what specifically you see under Linux? There is not really a driver to load. What happens if you run the eecutable from the command line?

stappersg, 2025-01-02:
On Thu, Jan 02, 2025 at 06:39:14AM -0800, Alex Taradov wrote:
> What do you mean by "low data level"?
> And what specifically you see under Linux?

Further follow-up should go in a fresh issue.
For the sake of "finding back".

Groeten
Geert Stappers
-- 
Silence is hard to parse

ataradov, 2025-01-02:
Why create more issues? This one is fine for followup.

stappersg, 2025-01-02:
On Thu, Jan 02, 2025 at 09:23:48AM -0800, Alex Taradov wrote:
> Why create more issues?

For finding back issue related to Wireshark.

> This one is fine for followup.

So here now more noise
that is **not** related to ".inf file invalid under windows 10"  :-/

Groeten
Geert Stappers
-- 
Silence is hard to parse

ataradov, 2025-01-02:
Your comments are also not related to anything. I'm fine with reusing the issues. There is not a problem with overwhelming amount of requests. If that happens, we'll figure out what to do.

closed 2026-10-10 by ataradov