ataradov/usb-sniffer #53: Wireshark data unreadable & Linux executable not working

WillemOuwerkerk, 2025-01-02:
Hi Alex,

This log i get in Wireshark in Windows:

1    0.000000000            Syslog    30    Starting capture
2    0.000000000            Syslog    31    Trigger input = 1
3    0.000000000            Syslog    21    VBUS ON
4    0.000000000            Syslog    40    Detected speed: Full-Speed
5    0.000000000            Syslog    37    Line state: J (3.50 ms)
6    0.003504066            Syslog    46    Line state: J [single] (1.20 us)
7    0.003505266            Syslog    39    Line state: J (998.77 us)
8    0.004504033            Syslog    46    Line state: J [single] (1.20 us)
9    0.004505233            Syslog    27    Line state: J
10    0.018503533            Syslog    46    Line state: J [single] (1.08 us)
11    0.018504616            Syslog    27    Line state: J
12    0.033503050            Syslog    46    Line state: J [single] (1.08 us)
13    0.033504133            Syslog    37    Line state: J (4.00 ms)
14    0.037502950            Syslog    46    Line state: J [single] (1.02 us)
15    0.037503966            Syslog    37    Line state: J (4.00 ms)
16    0.041501500            Syslog    37    Line state: K (1.43 us)
17    0.041502933            Syslog    27    Line state: J
18    0.059502116            Syslog    46    Line state: J [single] (1.08 us)
19    0.059503200            Syslog    27    Line state: J
20    0.076501083            Syslog    46    Line state: J [single] (1.35 us)
21    0.076502433            Syslog    37    Line state: J (2.00 ms)
22    0.078501016            Syslog    46    Line state: J [single] (1.42 us)
23    0.078502433            Syslog    27    Line state: J
24    0.093500650            Syslog    46    Line state: J [single] (1.08 us)
25    0.093501733            Syslog    27    Line state: J

In Linux i can not run the file, althoug i made it an executable.
Should i do same renaming, because the file is named: usb_sniffer_linux

ataradov, 2025-01-02:
Describe what is connected to the target port and how it is connected to the host PC. And also what are the sniffer settings. And does the device under test work as expected?

This looks like noise, so it may be speed misconfiguration.

What do you mean you can't run it? It is a command line tool, so what happens when you run it in the terminal?

WillemOuwerkerk, 2025-01-03:
Hi,

There is a CDC device connected to the sniffer (Your USB-sniffe-lite) as test object. The sniffer settings are as in your example.

I did change the USB-sniffer settings in Wireshark from high-speed to full-speed and in Windows all is working now. I can log USB data. Very nice, thank you.

Now the issue is Linux only, it is not working at all there.
Which steps can i take to check what's going on.

[image]

WillemOuwerkerk, 2025-01-03:
It needs GLIBC_2.34 how do i add that as a Linux newbee

ataradov, 2025-01-03:
What exact error messages you get?

If pre-built binary does not run on your system, just build from the source code. Run "make" in the "software" directory and provided you have the necessary dependencies, you will get a binary that is built for your system.

If the system was not setup for building before, you will get errors about missing tools or libraries. Install those tools and libraries. Details depend on your distro.

WillemOuwerkerk, 2025-02-12:
Hi Alex,

After recompiling it works under Linux too.
Got the sniffer working under Win10 too.
It is a very helpfull tool debugging my driver.
Thank you for the good work.

Willem Ouwerkerk

closed 2026-10-10 by ataradov