ataradov/usb-sniffer #56: Logging of CDC data fails

WillemOuwerkerk, 2025-03-06:
Hi Alex,

I can log the SETUP process of my CDC interface perfectly.
When i do an attempt to log the data sent when the connection is setup i see nothing.
Can you give me a clue of how to do that?
My current system is Windows 10 & Wireshark using the sniffer directly.

Kind regards,
Willem.

WillemOuwerkerk, 2025-03-06:
Yes, here is my last one made. This version gives showed transmitted dat on earlyer logs...

[usb-xs-6c.pcapng.gz](https://github.com/user-attachments/files/19111945/usb-xs-6c.pcapng.gz)

WillemOuwerkerk, 2025-03-06:
Hi Alex,
Yes that is part of the startup message. This version sends the characters one by one.
But any keystrokes after that are not shown.
How can i remove the IN/NAK pairs, i can try to find it out myself if it's to much trouble.

ataradov, 2025-03-06:
In the settings there is an option to fold empty frames, set it. Although I see occasional messages that empty frames are folded. If you have it set to fold, then something strange is going on. If you have folding disabled, then those sporadic messages are likely a software bug and I'll have to look into it.

As far as messages not showing, I don't see how they can disappear like this. Do you actually see the characters on the other side? What happens when you close the port? Do you see set line state/coding frames? They should happen when the port is closed.

ataradov, 2025-03-06:
Ok, I think you have folding enabled, but your host is very good at scheduling transfers. Right now there is a limit of 128 packets per frame, and your host schedules 136, so the folding stops. 

If you can build the binary, then change FOLD_BUF_SIZE from 128 to 200 in capture.c and recompile it. Otherwise, I'll build new binaries later today.

This would explain why folding does not work and it will make the log much cleaner. But it would not explain the missing data. I think it is really missing, so if you actually see the bytes on the other side, but not in the log, we'll have to look deeper into that after we can get captures with folding working.

ataradov, 2025-03-06:
I've updated the binaries. Folding should work now.  Do a new capture. The logs will have less spam.

WillemOuwerkerk, 2025-03-07:
This is my last log

[usb-xs-06e.pcapng.gz](https://github.com/user-attachments/files/19123214/usb-xs-06e.pcapng.gz)

Can you tell me how to put folding from the wireshark call of the sniffer?

ataradov, 2025-03-07:
Is this with the new binary? Double check that it actually is. This really looks like the old behavior. The folding is enabled, since you can see messages about folding at the beginning, but most of the stuff is not folded because there are more than 128 packers between SOFs.

Do other things I described - capture closing of the port to see if you get set line state and line coding messages.

What is the device? Is this CDC implemented on an MCU or a serial to USB adapter?

I would try with just an unconnected USB-to-serial adapter just to get a more controlled test.

ataradov, 2025-03-07:
It should be "-s fs". The device is Full-Speed, so the log contains just noise.

WillemOuwerkerk, 2025-03-07:
Yes sorry, i just discovered that too. This one looks nice.

[usb-xs03e.zip](https://github.com/user-attachments/files/19129988/usb-xs03e.zip)

ataradov, 2025-03-07:
Ok, this looks seems to at least end at the same place at as the pcapcmd one with "FORTH FORTH EXTRA ONLY : FORTH ) OK.0
@)".

ataradov, 2025-03-07:
And folding now works as expected. You can see all the empty frames go away.

WillemOuwerkerk, 2025-03-07:
Yes Alex, this looks a lot better. Also the data communication is captured.
Do you have any idea why wireshark still uses the old sniffer version?

ataradov, 2025-03-07:
You probably did not put it at the right path. Go to Help->About Wireshark->Folders and check "Global Extcap path". This is the place where the binary should be located.

WillemOuwerkerk, 2025-03-08:
Hi Alex,

Done now, everything is working smootly.
Now a can study the data stream en try to improve & optimise the cdc-driver i use.

Thanks again.
Willem

ataradov, 2025-03-08:
Closing the issue then.

closed 2025-03-08 by ataradov