ataradov/usb-sniffer #66: CY7C68013A-56PVXCT is obsolete

haldark, 2025-07-26:
Hello,

CY7C68013A-56PVXCT is obsolete and out of inventory in all major reputed distributors (Mouser, Digikey, Farnell, Arrow). Infinion also does not mention a replacement part, which is kind of surprising.

Without this major part unavailable, this project will be impossible to create.

I would appreciate if anybody can point me to a footprint compatible replacement part.

Best Regards,

ataradov, 2025-08-04:
For now I just wait until some better devices comes along. CBM9002A will be here for a very long time, and I expect availability to increase as CY device goes away. Dumping data into PC is a thing that is useful in many projects, so I'm actively looking for a solution. I'm waiting for a good tool-less design, but this may not happen, and then there are a few options:
1. The most direct is probably CH569W. But it is subject to the same issue as CBM9002A - only available from China, so availability is unpredictable.
2. Just use FPGA. I have a Verilog implementation of the USB HS peripheral. This was actually don for V1 of this project. So, the whole system could be 2 x USB PHY, FPGA and either really small external MCU, or internal RISC-V core. Internal core is attractive, but takes up a lot of memory blocks, so requires bigger FPGA just for that. This is not likely to be tool-less. There are Gowin FPGAs with embedded hard ARM core, but those are rarely actually available on the market.
3. Newer MCUs finally start to include good parallel capture interface.  STMH5/H7 have PSSI peripheral, which seems to be good. I will be experimenting with it in the current project I'm working on. But unfortunately, STMH5/H7 need USB PHY to make USB HS work and this requires at least 100 pin package.
4. I know for sure that this can be made to with with Atmel SAM E70. It has parallel capture interface that I know can do 60 MHz capture. It will be toll-less too. But it needs 100 pin package, but at least no external PHY. 

There are plenty of options, I'm just waiting until CY/CBM device availability really becomes an issue, and may be by that time some of the things above become available.

flowswitch, 2026-01-11:
Considering both the FX2 obsolescence and bandwidth insufficiency in some HS cases (reported buffer overflow issues), migrating to a SuperSpeed host-side interface would be great. FTDI FT60x? Cypress FX3?

ataradov, 2026-01-11:
I will never use FTDI products and FX3 is not available in reasonable packages nor at a reasonable price.

The best solution here is WCH CH569, which would work, but I'm also not interested in USB SS designs.

I'm currently working on an HS replacement that will simplify things a lot. It is going to take a bit, since I'm taking tangents to try multiple things out.

ataradov, 2026-01-13:
Wireshark decoding is definitely slow, but it should not  cause data loss, since there is a lot of buffering.
And there should not be a lot of line state events past the initial device enumeration.